MCUdude/microUPDI #2: Pro Micro hardware modification clarification

hiduino, 2019-08-31:
Can you clarify the hardware modification?  It appears that pin-1 (in the picture example) of the regulator is the RAW input, not the 5V output pin.

MCUdude, 2019-08-31:
It is indeed connected to the VIN pin, However, since microUPDI requires a 5V 16 MHz Pro Mini, this means that the solder jumper (SJ1 in the [Pro Micro schematics](https://cdn.sparkfun.com/datasheets/Dev/Arduino/Boards/Pro_Micro_v13b.pdf) is shorted, and ~4.5V will appear on the VIN pin.

I chose that connection point because it's the closest to the AREF capacitor. However, you may choose any 5V point on the board if you like.

hiduino, 2019-08-31:
Yes, thanks for the clarification.  I think it will be safer to go with the 5V output, just in case someone decides to put in more than 5.5V on the raw input (not that anyone would but you never know).

MCUdude, 2019-08-31:
> I think it will be safer to go with the 5V output, just in case someone decides to put in more than 5.5V on the raw input (not that anyone would but you never know).

There's no reason to connect the Raw pin to anything when using the Pro Micro as a UPDI programmer. I'll leave a note that the Raw pin shouldn't be used.

MCUdude, 2020-10-19:
The latest release (v1.0.1) uses a modified firmware where the hardware modification is no longer necessary since AREF is connected to VCC internally instead

closed 2020-10-19 by MCUdude