MCUdude/microUPDI #4: Can't program ATTiny3216 under SpenceKonde/megaTinyCore

tgtakaoka, 2019-10-12:
As noted on Usage section of MCUdude/microUPDI README.md, setting Arduino IDE's programmer to Onboard Atmel mEDBG (xplain_mini) (megaTinyCore) doesn't upload a sketch to bare ATTiny3216.

I think `-cjtag2updi` option of `avrdude` should be `-cxplainedmini_updi`.
Should I modify `avrdude.conf` of megaTinyCore to include `xplainedmini_updi`.
Am I right?

```
Forcing reset using 1200bps open/close on port /dev/ttyACM0
/home/t2/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/home/t2/.arduino15/packages/megaTinyCore/hardware/megaavr/1.0.5/avrdude.conf -v -pattiny3216 -cjtag2updi -P/dev/ttyACM0 -e -Uflash:w:/tmp/arduino_build_880748/Blink.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/t2/.arduino15/packages/megaTinyCore/hardware/megaavr/1.0.5/avrdude.conf"
         User configuration file is "/home/t2/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyACM0
         Using Programmer              : jtag2updi
avrdude: jtagmkII_recv_frame(): timeout
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_recv_frame(): timeout
avrdude: jtagmkII_getsync(): sign-on command: status -1
```

tgtakaoka, 2019-10-12:
Ah, I should report this to SpenceKonde/megaTinyCore project, not here.

closed 2019-10-12 by tgtakaoka

MCUdude, 2019-10-12:
You're right, there is something wrong with the way megaTinyCore handle programmers.

Thanks for your contributions to the MegaCoreX and megaTinyCore projects! Your time and knowledge are highly valuable 😊