MCUdude/microUPDI #5: Clone microUPDI not working

EclipseBoom, 2019-12-12:
Hi Hans,

I built a clone PCB of your design except I changed the 5V/3.3V buffer to a 74LVC1T45 for the serial port (not using at this point).  I did the AREF modification on the Micro.   Flashed the uUPDI core code fine.

Arduino IDE won't program with the unit.  Atmel Studio 7.0 sees the UPDI programmer but only allows ATMega4809 chips.  If I try to ID the chip it sees my chip ATTiny816 but won't allow me to program it since it is not a 4809.

Also, tried FLASHing the Mega32u4 manually using the .hex files.  Same behavior.  Working on Windows 7 64-bit.  Any ideas?

MCUdude, 2019-12-13:
> I built a clone PCB of your design except I changed the 5V/3.3V buffer to a 74LVC1T45 for the serial port (not using at this point). I did the AREF modification on the Micro. Flashed the uUPDI core code fine.

Awesome, would you like to post a picture of it? The LVC1T45 is probably even better for this than LVC1G125, but I used this because I literally have a pile of them after getting loads of free components on rolls from a former college.

> Arduino IDE won't program with the unit. Atmel Studio 7.0 sees the UPDI programmer but only allows ATMega4809 chips. If I try to ID the chip it sees my chip ATTiny816 but won't allow me to program it since it is not a 4809.

Have you tried with Avrdude? Before I figured out the AREF mod it did work with Atmel Studio 7, but not with Avrdude.
When it comes to the actual firmware this is pulled off an Arduino Uno Wifi Rev2, because this was the newest mEDBG version I was able to retrieve (1.13). This chip is probably set up to only work with ATmega4809 by default since this should be the only target if used on an Uno Wifi Rev2. I don't remember exactly where, but somewhere in the settings, you can choose to show all "compatible" chips for the connected programmer, not just the default one. IIRC I did this on my work computer.

EclipseBoom, 2019-12-13:
Thanks for the pointer. In Atmel Studio under Options->Tools->Tool settings you can set the option "Hide unsupported devices" to FALSE and then the full list of AVR parts shows up.

The programming and verify works.  Yay!!!!

Attached are some images of my unit.

[image]

[image]

EclipseBoom, 2019-12-13:
Regarding the Arduino IDE.  When Onboard Atmel eEDBG ... (megaTinyCore) is chosen as the programmer and the same connection to the target is used.  I get the following error messages:

It looks to me like the AVRDude configuration is not correct since the ATTiny816 does indeed have a UPDI interface.

`Arduino: 1.8.10 (Windows 7), Board: "ATtiny3216/1616/1606/816/806/416/406, ATtiny816, 20 MHz, 1.8V, Disabled, Disabled, EEPROM retained, 0.55V, Enabled, PORTA (pins 0~3,14~17), TX:14, RX:15, SDA:8, SCL:9, SCK:10, MISO:11, MOSI:12"

Sketch uses 2438 bytes (29%) of program storage space. Maximum is 8192 bytes.
Global variables use 97 bytes (18%) of dynamic memory, leaving 415 bytes for local variables. Maximum is 512 bytes.
C:\Users\Chris\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\Chris\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\1.1.2/avrdude.conf -v -pattiny816 -cxplainedmini_updi -Pusb -Uflash:w:C:\Users\Chris\AppData\Local\Temp\arduino_build_433361/ircomm_test.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Chris\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\1.1.2/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATtiny816
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash          0     0     0    0 no       8192   64      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        128   32      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel AVR XplainedMini in UPDI mode
         ICE hardware version: 0
         ICE firmware version: 1.19 (rel. 57)
         Serial number   : MICROUPDIPROGRAMMERX
         Vtarget         : 5.00 V
         JTAG clock megaAVR/program: 0 kHz
         JTAG clock megaAVR/debug:   0 kHz
         JTAG clock Xmega: 0 kHz
         PDI clock Xmega : 0 kHz

avrdude: jtag3_initialize(): part ATtiny816 has no UPDI interface
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

EclipseBoom, 2019-12-13:
I think I got it.  In avrdude.conf  at line 15210 there is:

```
part
    id		= ".avr8x";
    desc	= "AVR8X family common values";
#    has_updi	= yes;
    has_pdi	= yes;
    nvm_base	= 0x1000;
#    ocd_base	= 0x0F80;

```
Change to:

```
part
    id		= ".avr8x";
    desc	= "AVR8X family common values";
    has_updi	= yes;
#    has_pdi	= yes;
    nvm_base	= 0x1000;
#    ocd_base	= 0x0F80;

```
And it programs and verifies.  Yay!!! At least for the Tiny816 and Tiny402 chips I tried.  Don't know if this is a full solution - I'm not an avrdude.conf guy.

MCUdude, 2019-12-13:
Great looking hardware! Almost as good as mine! 😄 

Great to hear you figured it out. If you believe there's a bug in the megaTinyCore, you should definitely report it there, so it can be fixed properly.

closed 2019-12-13 by MCUdude

MCUdude, 2019-12-14:
I was just thinking about how Atmel Studio could identify that this particular mEDBG firmware should be default only be able to use used with ATmega4809.

After a bit of investigation it turns out that the 4809 device signature (1E 96 53) is actually [burned into the EEPROM](https://github.com/MCUdude/microUPDIcore/blob/f5c20418ca8572356122b39f8d53cf9c319c58b7/avr/firmwares/mEDBG_UPDI_1.13_modified_suffer.eep#L10). I don't have time to test this theory at the moment, but I think Atmel studio will detect it as a generic mEDBG if these three bytes was set to 00 00 00 or FF FF FF. All you need to do is to change the values in the .eep file and calculate the 8-bit checksum at the end.

mraardvark, 2019-12-16:
The information in the EEPROM on the mEDBG is only for Studio to populate the device selection drop-down.  If you go into Studio options and un-check "hide unsupported devices" then the full device selection list is made available.

EclipseBoom, 2019-12-16:
> The information in the EEPROM on the mEDBG is only for Studio to populate the device selection drop-down. If you go into Studio options and un-check "hide unsupported devices" then the full device selection list is made available.

Ideally what we want is for Atmel Studio to auto populate that dropdown with all UPDI programmable devices.

EclipseBoom, 2019-12-16:
> I was just thinking about how Atmel Studio could identify that this particular mEDBG firmware should be default only be able to use used with ATmega4809.
>
> After a bit of investigation it turns out [...]

So I tried changing these to all 00 00 00 or FF FF FF with the correct checksum and there is no change.  Studio still only offers the Mega4809 with "Hide Unsupported" set as TRUE.  Bummer.

Note, the tool does correctly find the chip that is attached.  An reading the device sig does work.  But with "Hide unsupported" as TRUE the software won't enter programming mode.