MCUdude/microUPDI #7: working with Platformio

paulskirk53, 2021-03-13:
Hi, I have the microUPDI, can I get it to debug code on an AVR 4809 using Platformio? Any help to configure would be much appreciated.

maxgerhardt, 2021-03-14:
Per topic in https://community.platformio.org/t/debug-an-avr-4809-with-microupdi/19926/18?u=maxgerhardt it worked by adding 

```ini
upload_protocol = custom
upload_flags =
    -C
    $PROJECT_PACKAGES_DIR/tool-avrdude-megaavr/avrdude.conf
    -p
    $BOARD_MCU
    -c
    xplainedmini_updi
    -Pusb
    -b115200
upload_command = avrdude $UPLOAD_FLAGS -U flash:w:$SOURCE:i
``´

to the `platformio.ini`, after seeing the info in https://github.com/MCUdude/microUPDI#usage.

MCUdude, 2021-03-20:
> Hi, I have the microUPDI, can I get it to debug code on an AVR 4809 using Platformio? Any help to configure would be much appreciated.

PlatformIO doesn't support AVR debugging natively. You'll have to use Atmel Studio, MPLAB X, set up your own GDB to do this or follow the guide that @maxgerhardt linked to.

closed 2021-03-20 by MCUdude