MCUdude/microUPDI #9: Have you ever got your microUPDI working on MacOS Big Sur?

tombull, 2021-04-30:
If so, how? For me, avrdude will not communicate with the microUPDI (or any other mEDBG / nEDBG devices) on Big Sur. Most often get something like `usbdev_open(): no usable interface found` or another `usbdev_open()` error, and I've not found any way to get it to work.

tombull, 2021-04-30:
(Note: this is really not an issue with microUPDI - if anything, it's an issue with MacOS Big Sur... so whatever the answer is, please feel free to close this issue!)

MCUdude, 2021-04-30:
Hi!

Nope, I haven't tried with Big Sur. I'm planning to wait as long as I can before installing it because it's known to break lots of stuff like 32-bit support. You could try to install MPLAB X and see if it provides drivers Avrdude can use. I'll also recommend you to ask over at the Avrfreaks forum. I'm sure some of the gurus over there know.

tombull, 2021-04-30:
Thanks, I've got MPLAB X installed, but avrdude still doesn't like communicating with any programmers apart from JTAG2UPDI, sadly. For reference, if anyone in the future finds this with the same issue, I posted on Avrfreaks here: [MacOS Big Sur & avrdude with mEDBG, nEDBG or ATMEL-ICE](https://www.avrfreaks.net/forum/macos-big-sur-avrdude-medbg-nedbg-or-atmel-ice).

closed 2021-04-30 by tombull