MCUdude/microUPDI #11: connect 4808

yourSpace24, 2021-08-02:
Hi, i was able to program the pro micro

if i use arduino to check after the upload the programmer i got following data:

BN: Arduino Uno WiFi Rev2
VID: 03eb
PID: 2145
SN: MICROUPDIPROGRAMMERX

then i hook it up to a 4808 atmel and try to upload the firmware via arduino and get following error:

Global variables use 527 bytes (8%) of dynamic memory, leaving 5617 bytes for local variables. Maximum is 6144 bytes.
avrdude: usbdev_open(): WARNING: failed to set configuration 1: Device or resource busy
avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

An error occurred while uploading the sketch

the programmer is connected via ground and updi to the atmel 4808 chip it is a brand new one. I checked with an oscilloscope the updi pin to see what it is going on. An some bits are showing up.

could it be that it is a permission problem on the linux machine or is there a way to check a kind of simple communication to verify that everything is okay. The atmega 4808 runs on 3.3 volt.

many thanks

cheers

yourSpace24, 2021-08-02:
avrdude -v -patmega4808 -cxplainedmini_updi -Pusb -b115200 -e

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/root/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
   Using Programmer              : xplainedmini_updi
         Overriding Baud Rate          : 115200
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega4808
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         64   64      0     0     0 0x00 0x00
           flash          0     0     0    0 no      49152  128      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel AVR XplainedMini in UPDI mode
         ICE hardware version: 0
         ICE firmware version: 1.19 (rel. 57)
         Serial number   : MICROUPDIPROGRAMMERX
         Vtarget         : 5.00 V
         JTAG clock megaAVR/program: 0 kHz
         JTAG clock megaAVR/debug:   0 kHz
         JTAG clock Xmega: 0 kHz
         PDI clock Xmega : 0 kHz

avrdude: Short read, read only 0 out of 64 bytes
avrdude: jtag3_edbg_recv(): Unexpected response 0xf0
avrdude: retrying with external reset applied
avrdude: Failed to write 64 bytes to USB
avrdude: jtag3_edbg_send(): failed to send command to serial port
avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x00
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: jtag3_edbg_signoff(): unexpected response 0x81, 0x11
avrdude: jtag3_edbg_signoff(): unexpected response 0x01, 0x00

avrdude done.  Thank you.

MCUdude, 2021-08-02:
Are you using a microUPDI programmer, or is is a breadboarded pro micro? What resistor value do you use in series on the UPDI line? Are you able to run the 4808 on 5V, just to check?

yourSpace24, 2021-08-02:
i use the pro micro with the micro updi firmware

yourSpace24, 2021-08-02:
with the 330Ohm resistor on the updi pin

yourSpace24, 2021-08-02:
the 4808 runs on its own LDO 3.3 volt i connect only the GND and updi to the programmer runs on 5v

yourSpace24, 2021-08-02:
Here the full avrdude comand i always get different device signatures
<pre> 
root@HAL9000:~# /home/theurlm/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/bin/avrdude -C /home/theurlm/.arduino15/packages/MegaCoreX/hardware/megaavr/1.0.8/avrdude.conf -v -patmega4808 -cxplainedmini_updi -Pusb -Uflash:w:/home/theurlm/Arduino/PV.MUX.200_REV.A/PV.MUX.200_REV.A.ino_atmega4808_16000000L.hex:i  -b8 -F

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/theurlm/.arduino15/packages/MegaCoreX/hardware/megaavr/1.0.8/avrdude.conf"
         User configuration file is "/root/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
         Overriding Baud Rate          : 8
avrdude: usbdev_open(): Found mEDBG CMSIS-DAP, serno: MICROUPDIPROGRAMMERX
avrdude: usbdev_open(): WARNING: failed to set configuration 1: Device or resource busy
avrdude: max packet size expected 912, but found 64 due to EP 0x81's wMaxPacketSize
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega4808
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         64   64      0     0     0 0x00 0x00
           flash          0     0     0    0 no      49152  128      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel AVR XplainedMini in UPDI mode
         ICE hardware version: 0
         ICE firmware version: 1.19 (rel. 57)
         Serial number   : MICROUPDIPROGRAMMERX
         Vtarget         : 5.00 V
         JTAG clock megaAVR/program: 0 kHz
         JTAG clock megaAVR/debug:   0 kHz
         JTAG clock Xmega: 0 kHz
         PDI clock Xmega : 0 kHz

avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x70a469
avrdude: Expected signature for ATmega4808 is 1E 96 50
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.

avrdude done.  Thank you.

</pre>

MCUdude, 2021-08-02:
Can you try to connect the AVCC pin to 5V on the Pro Micro?

[image]

The reason for this is that the original mEDBG firmware requires to do this in hardware. However, the microUPDI hex file has been modified, so AVCC should now be connected to VCC internally. But you should still try this mod to check if it makes a difference.

yourSpace24, 2021-08-03:
hi, same result i tried as well with the atmega 4808 on 5volt same result something is not working. Is there a way to check if the programmer is working correct ?

yourSpace24, 2021-08-03:
Hi again, ok made a new PCB added a new atmega 4808 on and now it is working (5v). Maybe the other one got fried. Just to be sure is the 3.3 volt target system okay ? or could be this a issue. many thanks

MCUdude, 2021-08-03:
Well, I have a test board for the ATmegaXX08 chips where I can set the target voltage to 3.3V. I'm able to upload to it just fine using a microUPDI programmer. So I don't think a 3.3V system should be an issue. You could try to change the series resistor to something higher, for instance, 470R or 1k