MCUdude/microUPDI #12: Debug and serial monitor

paulskirk53, 2021-08-28:
If I use my microUPDI as a debug tool in MCSudio, it works fine. However, as part of the debug session, I'd like to send and receive chars over USART on my AVR4809. Currently I'm trying to do the serial comms using the Arduino sermon, but the problem I have is when debug is active in MCStudio, the Arduino Sermon is inactive. If I stop the debug session in MCStudio, the Arduino Sermon becomes active again. So my question is - Is it possible to run the debug session and serial data transmit/ receive simultaneously?
thanks.

MCUdude, 2021-08-28:
I haven't really tried to use it as a debugger while also having an active Serial monitor. Does the COM port disappear from the Windows device manager when you start a debug session? If so, I don't think you can do both at the same time.

paulskirk53, 2021-08-28:
Interesting, it does work, the COM port is available during debug. I have a live debug session in Microchip studio and a live Serial Monitor in arduino IDE. 
What doesn't seem to work (and what prompted my original post), is if I set a breakpoint with an action to log a message and variable value in the output window and check the 'continue execution box'. In that configuration, the serial monitor is inactive.

I don't think its a problem with the microUPDI 

thanks for reply.

MCUdude, 2021-08-28:
Thanks for the explanation! You could try a different terminal program, such as Realterm to figure out if it is the programmer's fault or not

paulskirk53, 2021-08-28:
I tried it with X-CTU as well, and it's possible to debug in MCStudio and simultaneously send and receive chars over X-CTU. Nice.

thanks.

MCUdude, 2021-08-28:
Great to hear you figured it out!

closed 2021-08-28 by MCUdude