MCUdude/microUPDI #13: Printable case for the microUPDI

egeltje, 2021-08-28:
I have just uploaded a printable case for the microUPDI to thingiverse. I find it useful and perhaps others might find it useful too.
https://www.thingiverse.com/thing:4943607

MCUdude, 2021-08-28:
Thanks for sharing, that is a quite nice case indeed!

To my surprise, the microUPDI has turned out to be the most popular board I sell on Tindie, even though there are many good UPDI programmer alternatives out there.
Few people also leave feedback, so I feel like I should ask; What made you buy the microUPDI and not all the other alternatives out there like the JTAG2UPDI, Microchip SNAP, PICkit4? Was it the price point, functionality, or software support? 🙂
Anyways, I hope you're happy with yours!

egeltje, 2021-08-30:
I have created one (http://fabacademy.org/2021/labs/waag/students/kooi-erwin/report/week_04/#project-updi) of the 4k7 resistor ones (with a ft2301 serial converter) and pyupdi tool but that gave sometimes finicky results. As I have fallen in love with the new iteration of the AVR family, it was time for a more solid programmer.
And to somewhat support your work on the Arduino code for supporting these uC's. :-)

merlinblack, 2022-01-13:
> And to somewhat support your work on the Arduino code for supporting these uC's. :-)
That's also why I bought one of your boards too :-) And your Mighty, Micro, Mega, etc cores too.

platima, 2024-11-13:
> I have just uploaded a printable case for the microUPDI to thingiverse. I find it useful and perhaps others might find it useful too. https://www.thingiverse.com/thing:4943607

Nice one! Do you mind if I 3D print these en-mass and list them in my store to buy? I'd just be charging for the time & material, not your design, and of course link to your files!

If not, all good too! Just trying to stock more cases these days, as there are not enough!