MCUdude/microUPDI #14: Custom UPDI programmer and ATmega4809

ITstreet1, 2022-01-24:
I tried to build a stand-alone programmer and a simple board with 4809 but something I didn't do good enough.

UPDI programmer
This programmer I made according to the schematic of an add-on board and the schematic of the Sparkfun Pro Micro 5V.
This programmer is alive and while following the steps here, I flash it, and it goes fine. My PC recognizes it as mEDBG Virtual Port. This part should be ok.

[image]

As you can see, there is not much. Only what needs to be.

ATmega4809 board
I followed the barebone schematic here as well. Just solder all the necessary things, but when trying to upload a basic Blink sketch I get errors.

[image]

On this second board, I didn't solder the 3.3V regulator, LEDs, etc. Only the basic.

I choose MegaCoreX, Atmega4809
Internal clock 16MHz
BOD 2.6V
48 pin standard
Reset pin: Reset
No Bootloader
Programmer mEDBG/microUPDI

To upload I use Sketch->Upload using programmer

How to troubleshoot this? From where to start?
Are my wirings ok?
Did I choose the right options in Arduino IDE?
Do I need a step or two before uploading the sketch?

Thank you in advance.

MCUdude, 2022-01-24:
Are you able to program your ATmega4809 board with a different UPDI programmer, for instance, a JTAG2UPDI? And which error are you getting from Avrdude?

ITstreet1, 2022-01-24:
I just saw I didn't connect PF0 and PF1 to the UPDI connector as an RX and TX

MCUdude, 2022-01-24:
> I just saw I didn't connect PF0 and PF1 to the UPDI connector as an RX and TX

Did this solve your issue?

ITstreet1, 2022-01-24:
This is what I get:

```
avrdude: Short read, read only 0 out of 64 bytes
avrdude: jtag3_edbg_recv(): Unexpected response 0x90
avrdude: retrying with external reset applied
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: retrying with external reset applied
avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: jtag3_edbg_signoff(): unexpected response 0x81, 0x11
An error occurred while uploading the sketch
avrdude: jtag3_edbg_signoff(): unexpected response 0x01, 0x00
```

ITstreet1, 2022-01-24:
I cut the wrong RX/TX traces and rewire them. Error as above.

I have no idea what this means, except I have to check my connections.
Connections are good. Now. :) 
I could mess a thing or two with programmer, 4809 board or both

MCUdude, 2022-01-24:
I have sold _many_ microUPDI boards, and as far as I know, none of the buyers has had any issues getting it to work.

The problem with your setup is that you have two potential sources of error, the programmer and the board. Get yourself a programmer you _know_ works, for instance, a DIY JTAG2UPDI programmer using an ATmega328 based board such as an Arduino UNO or Nano.

ITstreet1, 2022-01-24:
OK.
As far as I can see, it uses the UPDI line only.
Another thing, I have to disable the reset. The easiest way is to use a Pro Mini, 5V. I can disconnect the DTR pin.
Anything else I should have to do?

merlinblack, 2022-02-21:
With my microUPDI I'm seeing the same problem with avrdude compiled from source, but not with the avrdude included with the Arduino IDE (1.8) on Fedora. Works really well with Atmel Studio 7 in windows too.

I want to get avrdude from source as the Arduino IDE one is kinda weird in other ways, and also I want to compile for a raspberry pi.

It's quite possibly my less-than-great soldering, but it's weird it always works for some ways, but not others.

Possibly related is that I get gibberish serial comms as if the baud is wrong via the microUPDI, but the serial from 4809 works via another serial to USB I have (set at 115200).

```
avrdude: Version 6.99-20220211 (7b79b72)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/home/nigel/prog/avr/avrdude/build_linux/src/avrdude.conf"
         User configuration file is "/home/nigel/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega4809
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
           osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           usersig                 0     0     0    0 no         64   64      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      49152  128      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel AVR XplainedMini in UPDI mode
         ICE HW version  : 0
         ICE FW version  : 1.19 (rel. 57)
         Serial number   : MICROUPDIPROGRAMMERX
         Vtarget         : 3.30 V
avrdude: Short read, read only 0 out of 64 bytes
avrdude: jtag3_edbg_recv(): Unexpected response 0x58
avrdude: retrying with external reset applied
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: retrying with external reset applied
avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80

avrdude done.  Thank you.
```