MCUdude/microUPDI #16: Question : Can I use 3.3V Pro Micro

hallard, 2022-07-13:
If I need to program only 3.3V target, can I use 3.3V Pro micro (I have only this version in hand) or does the firmware going into the pro-micro (8MHz) will not work due to different speed? 
I'm pretty sure it will not work until FW is build for 8MHz but worth asking

platima, 2024-11-13:
It would likely work in some scenarios, as UPDI should be fairly tolerant of speed changes and I think operates at way less than 8MHz (0.25-1MHz I think), but may have bugs.

You'd need to change 
`microupdi.build.f_cpu=16000000L`
In boards.txt

I don't have Altium though, so I am unsure if microUPDI circuitry taps off RAW, which would still allow you to use 5V targets, else if it uses VCC, you'll be limited to 3.3V targets.

@MCUdude can probably confirm this!

Else, just try it and find out 😊 I don't see it harming anything.

MCUdude, 2024-11-14:
The original EDBG programmers all ran at 16 MHZ and 5V. But feel free to try it. I'm not actually sure if the AVR uses the CLKPR register to clock the 16 MHZ clock down to 8 internally.

closed 2024-11-17 by MCUdude