MCUdude/microUPDI #17: Firmware upload fails

paulskirk53, 2023-05-06:
Hi, I'm using the arduino IDE 1.8.13 and can't get the firmware to upload.
My arduino pro micro has a reset pin and I tried grounding that as the upload started, but it has no effect. Anything else I can try please?

error msg from IDE:

Arduino: 1.8.13 (Windows 10), Board: "microUPDI Firmware Uploader"

Sketch uses 0 bytes (0%) of program storage space. Maximum is 28672 bytes.

Global variables use 0 bytes (0%) of dynamic memory, leaving 2560 bytes for local variables. Maximum is 2560 bytes.

Couldn't find a Board on the selected port. Check that you have the correct port selected.  If it is correct, try pressing the board's reset button after initiating the upload.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

paulskirk53, 2023-05-07:
I tried using a Uno as ISP with SPI and gnd pins connected to the pro micro, with both devices powered, in order to burn a bootoader for the pro micro. It didn't work, so I also used avrdude from a windows command line with -F as below, but no success, just get 'invalid device signature' when burning the bootloader. Any thoughts much appreciated.

C:\Users\Paul\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\Paul\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf -v -F -patmega32u4 -cstk500v1 -PCOM24 -b19200 -e -Ulock:w:0x3F:m -Uefuse:w:0xCB:m -Uhfuse:w:0xD8:m -Ulfuse:w:0xFF:m

paulskirk53, 2023-05-07:
ok so I got the sparkfun bootlaoder loaded (I missed out pin 10 of the arduino programmer which resets the pro micro), but the pro micro does not present a COM port in the arduino IDE. So I have tried :
-reset twice within 0.75 seconds to set the bootloader timeout to 8 seconds
-reset as I upload a sketch
but no port becomes available.
thanks for help with this.

paulskirk53, 2023-05-07:
looks like it was a faulty USB cable...:( I now have a port assigned to the pro micro using the reset immediately before upload technique.

paulskirk53, 2023-05-07:
ok, now finally got mEDBG virtual com port in windows device manager. The board has to be reset manually (two wires and a pushbutton switch temporarily installed onto pro micro rst and gnd) before upload even when the pro micro port is selected.
It's worth knowing that the standard sparkfun bootloader timeout is 750mS, which can be changed to 8 seconds by resetting the board twice within the 750 mS. It makes it much easier later on when the pro micro has to be reset when uploading the firmware. Good luck :)

paulskirk53, 2023-05-08:
Ok so I made good progress, but the code upload just hangs. Using Arduino ide, the tools menu has these items in it when the microUPDI is connected:
Board ATMega4809
clock internal 16Mhz
BOd 2.6V
Pinout "40 pin standard"
reset pin "Reset"
Bootloader no bootloader
Port "COM10 (Arduino Uno wifi Rev 2)"
get board info
Programmer "Atmel mEDBG (ATmega32U4)"
Burn bootloader

The IDE just contains empty setup() and Loop() functions

When I click the upload icon in the ide, the message window shows the following and the process just hangs indefinitely.

Any help much appreciated

Sketch uses 510 bytes (1%) of program storage space. Maximum is 49152 bytes.
Global variables use 4 bytes (0%) of dynamic memory, leaving 6140 bytes for local variables. Maximum is 6144 bytes.
C:\Users\Paul\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\Paul\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.7/avrdude.conf -v -patmega4809 -cxplainedmini_updi -Pusb -Uflash:w:C:\Users\Paul\AppData\Local\Temp\arduino_build_12643/sketch_may08a.ino.with_bootloader.hex:i -Ufuse0:w:0x00:m -Ufuse1:w:0x54:m -Ufuse2:w:0x01:m -Ufuse4:w:0x00:m -Ufuse5:w:0xC9:m -Ufuse6:w:0x06:m -Ufuse7:w:0x00:m -Ufuse8:w:0x00:m -Ulock:w:0xC5:m 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Paul\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.7/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi

when I unplug the USB cable, the message window shows:

avrdude: usbhid_open(): No response from device
avrdude: jtag3_open_common(): Did not find any device matching VID 0x03eb and PID list: 0x2145

avrdude done.  Thank you.

An error occurred while uploading the sketch

MCUdude, 2023-05-08:
Hi Paul!

> megaavr\1.0.7

Please download the latest MegaCoreX version. Avrdude 6.3 has been replaced with Avrdude 7.1, which should work much better with the microUPDI

paulskirk53, 2023-05-09:
Hi, I have installed avrdude 7.1 and its .conf file

I tried to reinsall the firmware, but I now get the output below from Arduino IDE, any help much appreciated.

Arduino: 1.8.13 (Windows 10), Board: "microUPDI Firmware Uploader"

Sketch uses 0 bytes (0%) of program storage space. Maximum is 28672 bytes.

Global variables use 0 bytes (0%) of dynamic memory, leaving 2560 bytes for local variables. Maximum is 2560 bytes.

Forcing reset using 1200bps open/close on port COM10

PORTS {COM1, COM10, } / {COM1, COM10, } => {}

PORTS {COM1, COM10, } / {COM1, COM10, } => {}

PORTS {COM1, COM10, } / {COM1, COM10, } => {}

PORTS {COM1, COM10, } / {COM1, COM10, } => {}

PORTS {COM1, COM10, } / {COM1, } => {}

PORTS {COM1, } / {COM1, } => {}

PORTS {COM1, } / {COM1, } => {}

PORTS {COM1, } / {COM1, } => {}

PORTS {COM1, } / {COM1, } => {}

PORTS {COM1, } / {COM1, COM10, } => {COM10, }

Found upload port: COM10

C:\Users\Paul\AppData\Local\Arduino15\packages\microUPDIcore\hardware\avr\1.0.2/scripts/avrdude-wrapper.bat C:\Users\Paul\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude C:\Users\Paul\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf atmega32u4 avr109 COM10 57600 C:\Users\Paul\AppData\Local\Arduino15\packages\microUPDIcore\hardware\avr\1.0.2 C:\Users\Paul\AppData\Local\Temp\arduino_build_456569 

avrdude: Version 7.1

         Copyright the AVRDUDE authors;

         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\Paul\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17\etc\avrdude.conf

         Using Port                    : COM10

         Using Programmer              : avr109

         Overriding Baud Rate          : 57600

         AVR Part                      : ATmega32U4

         Chip Erase delay              : 9000 us

         PAGEL                         : PD7

         BS2                           : PA0

         RESET disposition             : possible i/o

         RETRY pulse                   : SCK

         Serial program mode           : yes

         Parallel program mode         : yes

         Timeout                       : 200

         StabDelay                     : 100

         CmdexeDelay                   : 25

         SyncLoops                     : 32

         PollIndex                     : 3

         PollValue                     : 0x53

         Memory Detail                 :

                                           Block Poll               Page                       Polled

           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack

           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------

           eeprom                 65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00

           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00

           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00

           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00

           efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00

           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00

           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : butterfly

         Description     : Atmel for bootloader using AppNote AVR109

connecting to programmer: .avrdude butterfly_recv() error: programmer is not responding

avrdude butterfly_recv() error: programmer is not responding

avrdude butterfly_recv() error: programmer is not responding

avrdude butterfly_recv() error: programmer is not responding

avrdude butterfly_recv() error: programmer is not responding

avrdude butterfly_recv() error: programmer is not responding

Programmer id    = 0����; type = $

Software version =

paulskirk53, 2023-05-09:
In windows device manager the baord presents itself as expected:

[image]

but when I look in the tools > programmer menu, the only atmel option is Atmel ICE. The Atmel mEDBG (microUPDI) does not appear.

Not much hair left now :) :)

Would it be useful perhaps to erase the Pro Micro and start again from scratch?
thanks for any thoughts.

MCUdude, 2023-05-10:
Why would you want to reinstall the firmware?

The firmware that's currently on the AVR is not designed to play nicely with the built-in bootloader, so I've not been able to flash it one more time using the bootloader.

You can flash the firmware directly using Avrdude. Here's the command if you're using an Atmel ICE. I think this command should work. If not, manke sure the path to avrdude.exe, avrdude.conf and mEDBG_UPDI_1.13_no_bootloader.hex is correct.

```
C:\Users\Paul\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude.exe -C "C:\Users\Paul\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17\etc\avrdude.conf -catmelice_isp -patmega32u4 -Uflash:w:"C:\Users\Paul\AppData\Local\Arduino15\packages\microUPDIcore\hardware\avr\1.0.2/firmwares/mEDBG_UPDI_1.13_no_bootloader.hex":i
```

paulskirk53, 2023-05-10:
HI thanks for reply. I don't have an Atmel ICE. I have been through process as described for flashing the mEDBG firmware to the Pro Micro, but as I've posted, when I try to program my AVR4809 chip, the  Atmel mEDBG (microUPDI) does not appear in the IDE's programmer menu.

So where to go from here ? Any ideas much appreciated.

paulskirk53, 2023-05-10:
Also tried the programmer in Microchip studio and although it appears in the device programming tool dropdown, and is selectable, when 'Apply' is selected a 'failed to connect' error appears.