MCUdude/microUPDI #18: Onboard Atmel mEDBG option not present in programmers menu, 'initialization' fails during upload

Sk8rSeth, 2023-07-04:
i cant seem to upload anything to my attiny1616 through the microUPDI. i am new to UPDI programming so im not sure what im doing wrong.

my pro micro was programmed with the microUPDI firmware uploader correctly, and it now shows as 'Arduino Uno WiFi Rev2' under the com port menu.
however there is no option for the 'Onboard Atmel mEDBG' under the programmers menu (see below)
[image]

when i try to upload a blinky sketch via Upload Using Programmer, this is the output i get:

```
avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Mjolnir\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.8/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATtiny1616
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash          0     0     0    0 no      16384   64      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   32      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel AVR XplainedMini in UPDI mode
         ICE hardware version: 0
         ICE firmware version: 1.19 (rel. 57)
         Serial number   : MICROUPDIPROGRAMMERX
         Vtarget         : 3.30 V
         JTAG clock megaAVR/program: 0 kHz
         JTAG clock megaAVR/debug:   0 kHz
         JTAG clock Xmega: 0 kHz
         PDI clock Xmega : 0 kHz

avrdude: Short read, read only 0 out of 64 bytes
avrdude: jtag3_edbg_recv(): Unexpected response 0x18
avrdude: retrying with external reset applied
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: retrying with external reset applied
avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: jtag3_edbg_signoff(): unexpected response 0x81, 0x11
avrdude: jtag3_edbg_signoff(): unexpected response 0x01, 0x00

avrdude done.  Thank you.

Failed programming: uploading error: exit status 1```

seems that initialization has failed on the attiny1616, and i read a little (and understood even less) about the combination reset/updi pin on the attiny1616 and how there might be some kind of internal fuse i need to set or something to allow for programs to be uploaded? 

again im new to UPDI uploading and im not sure if there is like a reset window to program the attiny1616, or if i need to add a resistor to ground on one of the pins etc. OR if its even a problem with the attiny1616, my custom board not being soldered correctly (despite my inspection) or if the microUPDI programmer has some kind of error. is there any resistors i need to set on the microUPDI after uploading the UPDI firmware to the pro micro?

thanks in advance for any help

MCUdude, 2023-07-04:
I see you're using megaTinyCore, which bundles an outdated Avrdude (6.3) version with many bugs that were later resolved.

In order to figure out if you're facing a hardware issue, or if it's simply Avrdude 6.3 that doesn't work that well with the microUPDI, you should download Avrdude 7.1 from here:

https://github.com/arduino/avrdude-packing/releases/tag/7.1-arduino.1

Extract the archive, and run the following command (where you insert the path to the avrdude.conf file that came with Avrdude 7.1, and the path to the hex file you want to upload):

```
avrdude.exe -C /path/to/avrdude.conf -c xplainedmini_updi -U flash:w:path/to/file.hex
```

Sk8rSeth, 2023-07-04:
thanks for the quick reply. this command actually returns its own set of errors
```avrdude error: invalid file format '\Users\Mjolnir\Desktop\Projects\TestCode\TestCode.hex' in update specifier
avrdude 
error: unable to parse update operation 'flash:w:C:\Users\Mjolnir\Desktop\Projects\TestCode\TestCode.hex'```

i got the hex file by using `Sketch > Export Compiled Binary` from the Arduino IDE 1.8.19. i am assuming im using the avrdude command improperly but --help did not provide much on the command.

is there a way for me to replace the outdated avrdude version within megaTinyCore so i can try to upload from the IDE directly? im not familiar with the command line version of avrdude and im having a hard time finding documentation about the flags and commands.

MCUdude, 2023-07-04:
> this command actually returns its own set of errors

Try adding quotes around the path to the hex file, and add the `:i` at the end (meaning Intel hex file format)

`-U flash:w:"flash:w:"C:\Users\Mjolnir\Desktop\Projects\TestCode\TestCode.hex":i`

> Is there a way for me to replace the outdated avrdude version within megaTinyCore so i can try to upload from the IDE directly?

It's possible to replace the Avrdude binary megaTinyCore uses, and the avrdude.conf file, but if you can't figure it out, you can still use Avrdude directly without having to "export compiled binary" every time. Turn on verbose compilation and upload under the IDE settings, and you'll find the full path to where the compiled hex file is located after a successful compilation.

Avrdude 7.2 will soon be released, with a bunch of new features and improvements over 7.1, and especially over 6.3. Spence, the megaTinyCore maintainer, has said that he will update from 6.3 to 7.2 as soon as the release is ready.

Sk8rSeth, 2023-07-04:
Ah thank you, the quotes and the explicit intel format were the problem there. but i am getting the same problem as before:
failed to initialize. im not sure if this is a hardware error with this attiny1616 or something else. when it fails to initialize, do you know its trying to do? im wondering if my circuit might have a resistor blocking a reset pulse (or not blocking when it should) or something like that?

full error from the command line upload:

``` avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude done.  Thank you.```

MCUdude, 2023-07-04:
Can you post the complete Avrdude 7.1 output, but this time, add `-v` at the end? This enables verbose output, which _may_ reveal something

Sk8rSeth, 2023-07-04:
absolutely
```
avrdude: Version 7.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\Mjolnir\Downloads\avrdude_Windows_32bit\etc\avrdude.conf

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATtiny1616
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
           osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           userrow     usersig     0     0     0    0 no         32   32      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256   32      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      16384   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel AVR XplainedMini in UPDI mode
         ICE HW version  : 0
         ICE FW version  : 1.19 (rel. 57)
         Serial number   : MICROUPDIPROGRAMMERX
         Vtarget         : 3.30 V

avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude main() error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude done.  Thank you.
```

I am noticing that avrdude is specifying 3.3v with vTarget, but my system and attiny1616 are designed for 5v. the microUPDI is also using the 5v jumper. could that be the problem?

MCUdude, 2023-07-04:
> absolutely

Thank you! However, the verbose output doesn't reveal much more useful information. If you have a multimeter of some sort, can you check that there is a 330 ohm resistor between the UPDI pin on the 6-pin connector, and digital pin 7 on your Arduino Pro Micro?

> I am noticing that avrdude is specifying 3.3v with vTarget, but my system and attiny1616 are designed for 5v. the microUPDI is also using the 5v jumper. could that be the problem?

Most likely not. The Xplained Mini/microUPDI can't read the target voltage, and it doesn't care about it either. It runs at 5V no matter what. I'm not sure why the microUPDI firmware reports 3.3V, as I don't have the source code (only Microchip has, and they won't share it).

[image]

Sk8rSeth, 2023-07-04:
classic Microchip, not caring about open source work lol

my multimeter reads 331 ohm between those two pins, so i think its safe to say the resistor is in place.

i am starting to think it must be a problem with my circuit. im using the VQFN package of the attiny, so maybe there is a trace bridged underneath that i can visibly see.

Sk8rSeth, 2023-07-04:
> This may be it. I've used an ATtiny1616 in a VQFN package myself, and I had to reflow it twice + do some rework to get it to work as it should.

thats good to know, ill give this package more scrutiny. is this the kind of problem that one might see with a poorly reflowed QFN package? this is the first one ive used so im not sure how problems manifest with them yet

MCUdude, 2023-07-04:
My problem was that the chip "floated" on top of the solder, where some pads made contact while others didn't

Sk8rSeth, 2023-07-04:
thank you for all your help! glad to know the microUPDI is all good to go!

closed 2023-07-24 by Sk8rSeth