MCUdude/microUPDI #22: I'm stuck. Unable to flash the Arduino Pro Micro as microUPDI programmer

jaccobezemer, 2024-04-19:
Hello,

I have a working Pro Micro. Uploaded the Pro Micro Test Code and I get blinking LEDs.

When I select microUPDI Firmware Uploader from the boards menu and click the upload button I get an error.

```
The system cannot find the path specified.
exit status 1

Compilation error: exit status 1
```

What path can't be found? What am I missing?

jaccobezemer, 2024-04-20:
So I still don't know why I get the error.
But I managed to upload the firmware using a modified version of scripts\CLI flash script\Load_firmware.bat and a Arduino UNO as Arduino ISP.
It looks like a got a working microUPDI programmer as I managed to upload firmware to an ATMega4809.

closed 2024-04-20 by jaccobezemer

MCUdude, 2024-04-20:
Hi!
Sorry for the late reply! Great to hear you managed to upload the firmware through the alternative script provided. I'll try to look into why you weren't able to upload the firmware using the IDE.